minceheid/openeo #20: Home Assistant Control (MQTT)

MuddyRock, 2025-07-20:
Control via Home Assistant would be the dream.
One way to do this would be via MQTT.

Below is the formats of a config messages needed to create the Master On/Off switch in Home Assistant ("autodiscover" in Home Assistant - this saves having to enter the MQTT state topics etc into the yaml config).

I've also included some simple instructions on publishing a change of state of the switch (e.g. from ON to OFF) via MQTT so that Home Assistant can pick it up.

We would need similar config messages for each of the other options / settings and readings e.g. the Amps, the Mode, the read value for kW, volts, etc (although possibly, ,something on the Volts to only trigger if it changes by more than X - mine seems to constantly fluctuate between 239-241 and  I don't want to spam the MQTT server / HA).

I haven't yet worked out how to create config for the multiple choice values like Amps.

**Config message Master Switch**
Publish to topic: "homeassistant/switch/" + unique_id + "/config" where unique_id is a unique_id for the charger (I used my original EO serial number).

```
{
    "name": "Master Switch" ,
    "command_topic": "eohub/switch/EO_Hub_EO12345/master/set",
    "state_topic": "eohub/switch/EO_Hub_EO12345/master/state",
    "payload_on": "ON",
    "payload_off": "OFF",
    "state_on": "ON",
    "state_off": "OFF",
    "unique_id": "EO_Hub_EO12345_MasterSwitch",
    "device": {
        "name": "EO Hub EO12345",
        "identifiers": [
            "EO_Hub_EO12345"
        ]
    }
}
```

**Status message - Master Switch**
Publish ON or OFF (no quotes required to "eohub/switch/" + unique_id +"/master/state" (without the quotes!)

A similar message can be set up to create a sensor for the items which are readings only e.g. Volts. The important part is to ensure that each entity (sensor/switch) has a unique name, and that the message contains the "device" block with the same device name/identifier. This ties all the entities to the same device.

There may be some errors in the state topics above - I decided halfway through to rename some parts and might have missed one or two. I will attempt to double check later in the week.

I used my [Octopus Home Pro mqtt](https://github.com/MuddyRock/homepro_mqtt/blob/main/mqtt_publisher.py) as a starting point for the definition messages - this shows how you can have two sensors (but no switch) on the same device in Home Assistant.

tom66, 2025-08-01:
Hey @bfayers, any progress on MQTT?  It seems that it would be useful to have, but I don't want to start work on something like this if you're already halfway there.

tom66, 2025-08-31:
It's being actively discussed in the discord, but we're not quite sure on how to make it work nicely with the existing implementation.  Ideas welcome, I can send you an invite to the discord if you want to discuss with us.

tom66, 2025-08-31:
Sent.

joshbarbs, 2025-09-16:
This is critical for me too, so I created a fork and implemented it. I plan to raise a PR eventually. There is already a draft PR for this too.

tom66, 2025-09-16:
@joshbarbs Sounds like I should take a look at that before I end up writing something.  There's already an OCPP module but it can only be configured via JSON so the UI currently overrides its functionality.  I've just had a bit of writers block on how to integrate that with the present UI flow which works quite nicely for my application (I don't particularly need remote control but could imagine needing it in the future).

pnit, 2025-10-23:
I've also been @joshbarbs  fork for 3 weeks now with no hiccups.  I've been using it to limit charging power to my excess solar.

joshbarbs, 2025-11-06:
I've raised a PR for this at #84

MuddyRock, 2025-11-08:
Update: found the config options in libhomeassistant.py and have added the appropriate values to /home/pi/etc/config.json, but still not working (no MQTT messages showing in MQTT Explorer, and MQTT Enabled still showing as No in OpenEO frontend).

minceheid, 2025-11-08:
Ok - what I've done is 
1) deleted the 0.5.11 release - this means that people won't be installing it via the gui for the moment.
2) pushed the fix for get_user_settings() to main - I made a guess at what the range of values might be reasonable for publish_interval, so there may still be adjustments needed

So - for the moment - to get that fix that I've just pushed, just run:
``` openeo_download.py -r main```

MuddyRock, 2025-11-08:
While you were doing that, I spotted the change to homeassistant.py and downloaded the new version.
The good news is that the MQTT config options are now all in the database, and updating them in the front end updates them in the database.
Except: the frontend still shows homeassistant as not enabled, even though it shows as enabled in the configuration table in the database, and no MQTT messages are being sent.

I've now just also run 
`python3 openeo_download.py -r main`
and I'm getting the same results.

irata-coder, 2025-11-08:
In pluginParamSpec the default for publish_interval is 5, so it's outside the now UI range, i think we need:

```
util.add_simple_setting(self.pluginConfig, settings, 'slider', "homeassistant", ("publish_interval",), 'Publish Interval (seconds)', \
    note="Publish Interval (seconds)", \
    range=(5, 3600), default=5, step=1, value_unit="s")
```

Does it also need this added:

```
util.add_simple_setting(self.pluginConfig, settings, 'toggle', "homeassistant", ("enabled",), 'Enable Home Assistant MQTT', 
            note='Enable Home Assistant MQTT integration.')
```

As all other plugins (i think) look to all default to true for enabled in pluginParamSpec.  I'm not sure of this, as I haven't got my head around the codebase, maybe they work as they can't be disabled.

Maybe to test this theory temporarily change the pluginParamSpec inside homeassistant.py to ... {"type": "bool", "default": True} ...

I haven't installed on my EO Mini yet, so I can't test myself just yet.

minceheid, 2025-11-08:
If I have some time tomorrow, I'll take a closer look.

tom66, 2025-11-08:
IIRC, the 'enable' option for a plugin is automatic if the plugin isn't marked as a 'CORE' plugin that can't be disabled.

minceheid, 2025-11-09:
I think I see the problem. It's related to how the "enable" module option is added - this is the first "optional" module we've had in a long while, so it has exposed a flaw in that mechanism. I'm going to go and do some chores now, but will have a think about this and put a solution in later.

minceheid, 2025-11-09:
On the basis that it looks (to me) to be working, I've just republished this as release v0.5.11b

Notwithstanding any urgent bugs that get found over the next few days, this will be the last  release that I do until December.  I'm off to Thailand for a few weeks, and when I get back, the priority will be to focus on a UI refresh to enable multiple schedules to be defined.

zackslash, 2025-11-14:
@kwatt67 your MQTT Broker Host should be your homeassistant IP/hostname. Your screenshot show shows it set to 'localhost', that's possible (i guess) but unlikely.