minceheid/openeo #55: User "pi" is hard-coded

MuddyRock, 2025-08-20:
The deployed service file has the user "pi" hard-coded throughout (/home/pi), and deploy.sh has it hard-coded in one place (but $whoami in others).

This should be configurable - when I first wrote the OS to the SD card, I chose my own username, so the service failed to start.

minceheid, 2025-08-21:
This is a design decision, as the choice of username affects several things beyond the user itself (e.g. systemctl actions). The README.md makes it clear that the pi user must be used.

Particularly given that the RPi is likely to be dedicated to openeo as a single purpose, there are no plans to change this at this time to make that configurable. Whilst earlier (less complex) versions were a bit more flexible in this regard, the requirement to use the "pi" user was introduced as a consequence of increasing complexity, and the need to have a robust and consistent deployment mechanism. At the moment, I see no functional benefit in supporting a configurable username.

Be aware also that there is a new (more robust) deployment mechanism due to be pushed in coming days. It includes an explicit check that it is being run by the pi user.

minceheid, 2025-08-21:
Changing to "not planned"

closed 2025-08-21 by minceheid