minceheid/openeo #67: Occasional serial overruns/serial lockup

tom66, 2025-09-10:
I'm seeing a bug very rarely (once every few weeks?) where the serial comms will stop talking to the EO hardware and no UI updates are possible.  The UI still appears to work but nothing is live.

Restarting the service fixed the issue in at least one case, so I think the resolution is to ruggedise the serial reader/writer and add some handling for unexpected conditions, e.g. no comms in 10 seconds then kill the serial port and re-open it.  I've had experience with buggy Linux kernel drivers for serial ports on other embedded systems, so the chance of one on a Pi doesn't surprise me.

tom66, 2025-09-22:
Still occurring on 0.5.3.  Going to investigate when I can. Pressing 'restart openeo' in the settings page does fix it so presumably the serial code just needs better error handling.

minceheid, 2026-01-18:
Hi Tom, Nothing at all changed in the serial comms. Everything I've looked at recently has been front-end.

tom66, 2026-03-08:
Sorry for the delay.  This bug seems to occur about once every few days but since the fault is just to limit to 32A, I only notice when I lose the diagnostics and scheduler.  I'm seeing this in the log for this morning:

`Mar 08 03:05:53 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:05:58 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:05:59 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:04 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:05 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:10 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:11 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:16 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:17 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:23 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:23 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:29 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:29 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:35 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:36 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:41 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:42 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:47 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:48 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:53 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:54 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:06:59 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:00 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:05 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:06 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:11 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:12 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:17 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:18 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:23 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:24 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:30 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:30 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:36 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:36 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:42 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:42 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
Mar 08 03:07:48 charger openeo[643]: INFO - sendSerialCommand:81 - Response from serial was empty - possible serial overrun
[..]`

This is repeated throughout the log until a restart request, which fixes the issue.  My suggestion is that if `sendSerialCommand` is sent `N` times (say, 10) with no update or it returns None, then the port be re-initialised.  I'll see if I can submit a pull request for that later, along with some UI fixes for iPhone.