minceheid/openeo #72: Feature Request: Simple charging history log

chrisonhub, 2025-09-18:
The statistics page has a huge amount of technical information on and clearly very useful.  My request would be for a simple charging history page, so you can look over the last days/weeks and see a simple view of hours charged on each day.

It would be great to have a simple screen to look at and see "yep, it charged 5 hours last night" so I know my car is full and ready to go.  We have 2 EVs and one doesnt have an app to check its charge status.

minceheid, 2025-09-18:
Be aware that we already have session charging information on the home page:

[image]

This figure is reset when openeo detects to zero that the cable has been disconnected, so that should solve your immediate problem with the car that doesn't have an app. 

But that idea seems like a useful feature. At the point that we detect the cable being disconnected, we could log that session data to a persistent store, and show the history on a page.

minceheid, 2025-10-04:
Coming in the next release, which is in final testing at the moment...

[image]

minceheid, 2025-10-05:
Added in release 0.5.8

Note that there is an anomaly in the way that the sessions are recorded, and the summary charts are drawn. It records connection time, rather than charging time, so if you are using a timed schedule the session will assume that the charge is averaged over the entire connection time, which might give some odd results.

Example - I have my charging schedule set to 0200-0700, but I connected my car last night. My charts are showing that there was charging yesterday and today, which is incorrect. I might need to rethink/rework this aspect in due course.

chrisonhub, 2025-10-05:
Great looking graphs, will upgrade when I get chance later this week.  I think the EO app also recorded from 'connection time' which always confused me... I plug in 6pm and then charging starts 12.30am (the next day) but the charge is recorded against the first day.